nasa/PSP #10: PSP memory, port, and EEPROM functions assume direct-mapped access

skliper, 2019-09-25:
The PSP currently provides a number of access functions such as CFE_PSP_MemWrite8/16/32, CFE_PSP_PortRead8/16/32, etc.

These functions all assume that the memory is directly accessible to the current process by simply casting the address as a pointer and directly reading/writing from it.  Unfortunately this is often ***NOT*** the case.
* I/O port access in the x86/intel world is never memory mapped and requires different instructions (inline asm or kernel syscalls) in order to get to it.
* When using virtual memory, physical memory addresses are not directly accessible to the running process until the memory is mapped into the current virtual memory space.
* Many EEPROM devices are actually connected via a serial bus such as SPI or TWI and therefore would not be memory mapped.

With the way it is structured right now, cfe_psp_ram.c, cfe_psp_port.c, and (to a lesser degree) cfe_psp_eeprom.c only provide slow performance-robbing function calls to simply cast a pointer.

Furthermore, it is arguable whether direct I/O port or physical memory access even belongs in the PSP at all; the API this provides to the application layer remains far too hardware-specific to provide any useful abstraction.  Any CFS application performing direct I/O is already unlikely to be portable to any other platform, since (by definition) this would be accessing a specific hardware device at a specific location.

Proposed changes:
* Stop compiling cfe_psp_ram.c, cfe_psp_port.c, and cfe_psp_eeprom.c from the "shared" code by default; what these currently implement is more of an exception than the general rule.  These can be renamed or moved to indicate they are not always used.
* Deprecate/discourage future use of the RAM/port access functions.  Instead, a "driver" architecture should be used so the hardware device can be better abstracted (see nasa/ci_lab#6   This also allows easier simulation of the hardware, and the resulting CFS application will be much more portable/reusable.
* On the PSP's for which direct memory access //is// valid, they can continue to compile-in the current implementations to maintain backward compatibility.
* On the PSP's for which direct memory access //is not// valid, either customized functions can be provided or simply return the NOT_IMPLEMENTED error.

skliper, 2020-04-08:
@jphickey is this a candidate for config/source selection related updates?  I didn't refresh on all the comments, but separate the code and select source files as needed.

jphickey, 2020-04-08:
Yes - to clarify this would be a prime example for the fully-modular PSP concept.  With the pending merge that moves the startup code to OSAL BSP, this becomes more possible where a "PSP name" is just a collection of modular code abstraction blobs for that platform.  I'd like to see all remaining PSP functions, including these, go in that direction.

commit 29e8b7aa (2021-04-07): Fix #10, modularize the ram, port, and eeprom access

Convert the current "cfe_psp_ram.c" and "cfe_psp_port.c" routines
into modular components, and remove from "shared" dir.

The existing implementations become the corresponding "direct"
module, and are enabled based on the psp module selection.

Also added is a "notimpl" variant, where all the functions
return CFE_PSP_ERR_NOT_IMPLEMENTED.  This is used on Linux
or any other system where direct access is not possible.

Note this also renames the existing "eeprom_stub" module
to be "eeprom_notimpl" for consistency and to avoid any
confusion wi [... cut]
commit 5b572493 (2021-04-07): Merge pull request #282 from jphickey/fix-10-modularize-ram-port-access

Fix #10, modularize the ram, port, and eeprom access
PR #282 (2021-04-07): Fix #10, modularize the ram, port, and eeprom access
**Describe the contribution**
This is another attempt at resolving the long-standing issue of platform-specific code in the shared directory which is not easy to replace/override.

Convert the current "cfe_psp_ram.c" and "cfe_psp_port.c" routines into modular components, and remove from "shared" dir.  The existing implementations become the corresponding "direct" module, and are enabled based on the psp module selection.

Also added is a "notimpl" variant, where all the functions return CFE_PSP_ERR_NOT_IMPLEMENTED.  This is used on Linux
or  [... cut]
PR #289 (2021-04-08): psp Integration candidate: 2021-04-06
## Describe the contribution

### PR #286

Fix #171, use OSAL timebase for CFE timers

Cleans up stale code from the previous methods of generating 1Hz. Adds a new PSP module that instantiates an OSAL abstract timebase for use with cFE services. This single module is used across all psp implementations (mcp750, pc-linux, pc-rtems). Results in 1Hz timing tick on MCP750 will be more accurate. No changes to Linux or RTEMS

### PR #282 
Fix #10, modularize the ram, port, and eeprom access 

Fixes segfaults when `CFE_PSP_Port` routines are invoked on Linux. 

C [... cut]
closed 2021-04-08 by astrogeco