nasa/PSP #66: Use (foo), not &(foo[0]), where appropriate

skliper, 2019-09-25:
In a number of places, code jumps through extra hoops
to take the address of the first element of an array,
rather than just allowing the array name to gracefully
decay into the pointer.

Where just using the array name is more correct, we should
simplify the code to use it.

skliper, 2019-09-25:
_Trac comment by jhageman on 2019-07-08 09:45:22:_

Moved tickets that aren't ready for end of summer (2019) release to 1.4.1

skliper, 2019-09-25:
_Trac comment by jhageman on 2019-08-13 10:39:45:_

Did grep -r "[0])", "])" and "] )" just to catch most syntax, only cases were in fsw/mcp750-vxworks6.4/cfe_psp_memory.c:

{{{
*PtrToResetArea = (cpuaddr)&(CFE_PSP_ReservedMemoryPtr->ResetMemory[0]);
*PtrToUserArea = (cpuaddr)&(CFE_PSP_ReservedMemoryPtr->UserReservedMemory[0]);
*PtrToVolDisk = (cpuaddr)&(CFE_PSP_ReservedMemoryPtr->VolatileDiskMemory[0]);
}}}

Trivial fix...

zufishanali, 2019-11-27:
Hi @skliper, may I go ahead and make this change?

skliper, 2019-11-28:
@zufishanali Please do!  Note we are still trying to get the contribution process/documentation finalized with our legal office so it may take a while to merge (and there will likely be a form to sign), but any/all contributions are encouraged!  Likely you'll need to grep again to identify any current cases.

commit 2225fbfa (2020-02-25): Fix #66, Simplify address-of element 0
PR #138 (2020-02-28): IC - 20200226
Describe the contribution
Fix #66 

Testing performed
Steps taken to test the contribution:

    Current CI passes
    Test 4 in Enhanced CI in https://github.com/nasa/cFS/pull/40

Expected behavior changes
PR #116  - simplifies array handling in 'fsw/mcp750-vxworks/src/cfe_psp_memory.c'

System(s) tested on
    CI: Ubuntu 18.04

Additional context
N/A

Third party code
N/A

Contributor Info - All information REQUIRED for consideration of pull request
Gerardo E. Cruz-Ortiz - NASA/GSFC
closed 2020-02-28 by astrogeco