nasa/PSP #126: recommended lgtm issues:

avan989, 2019-12-20:
**Is your feature request related to a problem? Please describe.**
Recommended issue from lgtm:

cfe_psp_module.h
```
typedef enum
--
37 | {
  | In an enumerator list, the = construct should not be used to explicitly initialize members other than the first, unless all items are explicitly initialized.
38 | CFE_PSP_MODULE_TYPE_INVALID = 0,
39 | CFE_PSP_MODULE_TYPE_VALID_RANGE  = 1000,
```

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context about the feature request here.

**Requester Info**
Full name and company/organization if applicable

skliper, 2019-12-22:
Definitely worth clarifying.

https://github.com/nasa/PSP/blob/502e83ae179f0c439d6cddf310eede319eb44711/fsw/shared/cfe_psp_module.h#L36-L43

Should be understandable at a quick glance.  @jphickey any suggestions?  Not clear to me what the intent is here.

jphickey, 2019-12-31:
It is simply intended to provide unique(-ish) numeric values for the valid enumerators.  If every enumeration in the system starts with 1, then it is easy to interchange.  Using a starting value other than 1 offers a modicum of protection against using incorrect values (i.e. if someone passed a pointer to the wrong struct, or something).

I'm not sure why this is an lgtm "issue" as it is pretty well defined in the C standard how enum declarations work.  But if you'd rather start with 1, it won't change the behavior, but it will make it a little _less_ secure.

PR #235 (2021-01-25): Fix #126, Refactor CFE_PSP_ModuleType_t enum
**Describe the contribution**
Fix #126 - Avoids irregular enum warning

**Testing performed**
Build bundle and unit test (passed)

**Expected behavior changes**
None other than eliminates static analysis warning

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: Bundle main + this commit

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
commit 6d582be6 (2021-01-25): Merge pull request #235 from skliper/fix126-irregular_enum

Fix #126, Refactor CFE_PSP_ModuleType_t enum
commit c8dcf561 (2021-01-25): Fix #126, Refactor CFE_PSP_ModuleType_t enum

Avoids irregular enum warning
PR #243 (2021-01-27): psp Integration candidate: 2021-01-19
**Describe the contribution**

Fix #228, Refactor UT_ClearForceFail to UT_ClearDefaultReturnValue
Fix #126, Refactor CFE_PSP_ModuleType_t enum
Fix #236, Create Security Policy

**Testing performed**
See https://github.com/nasa/cFS/pull/177/checks

**Expected behavior changes**

PR #239 - Add `Security.md` with instructions on reporting vulnerabilities.

PR #229 - Rename `UT_ClearForceFail` as `UT_ClearDefaultValue` to reflect <https://github.com/nasa/osal/issues/724>

PR #235 - Remove unused elements in `CFE_PSP_ModuleType_t` enum to avoids irregular enum  [... cut]
closed 2021-01-27 by astrogeco