nasa/PSP #135: Implement psps without relying on aux_source_directory

matzipan, 2020-02-10:
**Is your feature request related to a problem? Please describe.**
The CMake documentation presents the following caveat for `aux_source_directory`:

> It is tempting to use this command to avoid writing the list of source files for a library or executable target. While this seems to work, there is no way for CMake to generate a build system that knows when a new source file has been added.

**Describe the solution you'd like**
The targets should have a list of files that go into making the specific psps. 

**Requester Info**
Andrei Zisu of Planetary Transportation Systems (Berlin, Germany).

jphickey, 2020-02-10:
I think this would be a good change, and it would make PSP's more modular and therefore more consistent with apps and the way everything else works.  With this the change to do an out-of-tree PSP is also more straightforward and logical.

jphickey, 2020-03-26:
This is actually necessary in order to merge nasa/osal#312, so I'll do this.

skliper, 2020-03-26:
Would this along with the related source selection via cmake options make the psp module concept redundant?
Or does the concept of adding modules still give us something unique?

jphickey, 2020-03-26:
I actually liked the idea of going full-modular with PSP's.  That is, the "psp name" simply translates to a list of modular components to build for platform support.  I briefly experimented with this and I think it could work nicely.

That is, in my experiment, the PSP was defined like this:

```
set(pc-linux_MODULELIST
    pc-linux-startup
    virtual-1hz
    no-exceptions
    no-ssr
    no-watchdog
    port_stub
    ram_stub
    eeprom_stub
)
```

This is convenient in that any/all of those modules can exist out-of-tree, using the currently existing module/app search logic.

skliper, 2020-03-27:
I definitely like the single mechanism for defining the modules included the PSP as you show above, and using the existing search logic!  Seems like a great simplification.

PR #148 (2020-04-14): Fix #135, Add separate CMakeLists.txt for each implementation
**Describe the contribution**
Do not use aux_source_directory to assemble a list of source files. Instead, put a proper CMakeLists.txt file in each implementation and build the implementation separately from the shared/common parts.

In addition to avoiding the aux_source_directory this allows PSP-specific compile definitions to be set on a per-implementation basis because it is defined separately.

Fixes #135 

**Testing performed**
Build code for all supported targets (ppc-vxworks6.9, i686-rtems4.11, native/x86-64 linux).  Verify  [... cut]
commit 2890a788 (2020-04-14): Merge pull request #148 from jphickey/fix-135-psp-separate-libs

Fix #135, Add separate CMakeLists.txt for each implementation
commit 34bd1819 (2020-04-14): Fix #135, Add separate CMakeLists.txt for each implementation

Do not use aux_source_directory to assemble a list of source files.
Instead, put a proper CMakeLists.txt file in each implementation
and build the implementation separately from the shared/common parts.

In addition to avoiding the aux_source_directory this allows PSP-specific
compile definitions to be set on a per-implementation basis because
it is defined separately.
closed 2020-04-20 by astrogeco