nasa/PSP #143: Fix travis-CI config warnings

skliper, 2020-03-25:
**Is your feature request related to a problem? Please describe.**
Build config validation (from travis-CI) -
root: deprecated key sudo (The key `sudo` has no effect anymore.)
language: unexpected sequence, using the first value (c)
root: missing os, using the default linux

**Describe the solution you'd like**
See https://github.com/nasa/ci_lab/commit/6cf5a57f5995710345b2a5411d11af2b2d7f80ee

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

astrogeco, 2020-03-27:
@skliper do you mind if we remove the "ready for CCB" label since the PR also has it?

skliper, 2020-03-27:
@astrogeco done, my mistake

PR #144 (2020-04-06): Fix #143, Resolve CI config warnings
**Describe the contribution**
Resolves CI configuration warnings
Fix #143 

**Testing performed**
Steps taken to test the contribution:
1. CI (CI change only)

**Expected behavior changes**
No Travis CI configuration warnings

**System(s) tested on**
 - Hardware: CI
 - OS: Ubuntu 18.04
 - Versions: this commit

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
commit 5fd08ace (2020-04-06): Fix #143, Resolve CI config warnings
commit a8e9709a (2020-04-06): Merge pull request #144 from skliper/fix143-resolve-ci-warn

Fix #143, Resolve CI config warnings
PR #151 (2020-04-09): Integration Candidate 2020-04-01
**Describe the contribution**
Integration Candidate
Fixes #143 

**Testing performed**
See PR #144 
Bundle CI - https://github.com/nasa/cFS/pull/64
Test Log: https://travis-ci.com/github/nasa/cFS/builds/159710339

**Expected behavior changes**
No Travis CI configuration warnings

**System(s) tested on**
Bundle CI - Ubuntu:Bionice

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman, NASA-GSFC
Gerardo E. Cruz-Ortiz, NASA-GSFC
closed 2020-04-09 by astrogeco