nasa/PSP #154: Inconsistent default CPU ID/Spacecraft ID between PC-Linux and other platforms

johnphamngc, 2020-04-21:
**Describe the bug**
The CPU ID/Spacecraft ID on Linux differs from the VxWorks build due to the Linux build deriving the value from the targets.cmake and other platforms deriving it from the value of [target name]_platform_cfg.h and [mission name]_mission_cfg.h

**To Reproduce**
Set CFE_MISSION_SPACECRAFT_ID in [mission name]_mission_cfg.h different than SPACECRAFT_ID in targets.cmake. Do same for CFE_PLATFORM_CPU_ID and the <x> value in TGT<x>. Build for Linux and a VxWorks target. CPU IDs will be different  when attempting to run SBN

**Expected behavior**
The same default CPU IDs should show up across both platforms.

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - OS: [e.g. Linux 4.4]
 - Versions cFE 6.7.12 , PSP 1.4 for mcp750, Linux, sp0

**Additional context**
Add any other context about the problem here.

**Reporter Info**
John Pham, Northrop Grumman

jphickey, 2020-05-15:
FYI - as part of the exception handling changes in pull #160 this also updated the way these IDs are handled.  The MCP750 implementation now matches the Linux and RTEMS in this regard.

skliper, 2020-05-15:
@johnphamngc Feel free to reopen if the update referenced above (now in master) didn't resolve your issue.

closed 2020-05-15 by skliper

johnphamngc, 2020-05-18:
Should a bug/enhancement be filed against cFE to deprecate CFE_PLATFORM_CPU_ID, CFE_PLATFORM_CPU_NAME, CFE_MISSION_SPACECRAFT_ID as they're now configured by TGT<x>_NAME, and SPACECRAFT_ID cmake variables?

skliper, 2020-05-18:
> Should a bug/enhancement be filed against cFE to deprecate CFE_PLATFORM_CPU_ID, CFE_PLATFORM_CPU_NAME, CFE_MISSION_SPACECRAFT_ID as they're now configured by TGT_NAME, and SPACECRAFT_ID cmake variables?

See nasa/cfe#710