nasa/PSP #180: Add replacement for deprecated OSAL - OS_IntLock and OS_IntUnlock

skliper, 2020-07-23:
**Is your feature request related to a problem? Please describe.**
OS_IntLock and OS_IntUnlock were deprecated from OSAL since there wasn't a platform independent way to implement.

**Describe the solution you'd like**
Need a replacement implemented where possible to support the concept of executing a uninterrupted block of code.

**Describe alternatives you've considered**
Non-portable implementation of uninterruptible code?

**Additional context**
Memory manager has a Load Memory with Interrupts Disabled command.  Perhaps operationally could be done differently?

*NOTE* - Priority to come up with a solution to support upcoming MM release.

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2020-07-23:
@ejtimmon @jphickey

jphickey, 2020-07-27:
As discussed previously we need to address what MM was (misguidedly?) trying to do with an IntLock.  The reason we deprecated `OS_IntLock` and `OS_IntUnlock` is because they don't have a consistent/well-defined effect.

Recommend to revisit the actual requirements for this command.  Note that saying "Interrupts Disabled" is really an implementation detail - not really an operational requirement.  That is, I would assume the requirement was to load memory in (some sort of) isolation, and disabling interrupts was the method of accomplishing that on some platform(s).

So the first step would be to rename the MM command to be something more appropriate - not reflecting implementation but the actual operational intent.  Then we can discuss how it can actually be implemented without OS_IntLock.

skliper, 2020-07-27:
@ejtimmon - can you help with the requirement and/or operational scenario here?  Or @acudmore, @jwilmot.

ejtimmon, 2020-07-27:
The relevant requirement seems to be MM2003:  “Upon receipt of a Write With Interrupts Disabled command, MM shall write up to <PLATFORM_DEFINED, TBD> bytes to the command-specified RAM memory address with interrupts disabled.”     Rationale: “Load with interrupts disabled so that can patch software that is executing.”

jphickey, 2020-07-29:
This seems pretty weak -- if you are patching code that is being actively executed, how does blocking interrupts help?  if the IP is already somewhere in the middle of the block when the command is received, it will have executed half old code and half new code, thereby putting it into a completely indeterminate state once the patch is applied.

One can certainly do it -- but it seems like a roll of the dice / hail mary pass sort of thing, to be attempted only when all other possible fixes have been tried and failed already.

Similarly, doing the `memcpy` in a high priority task (higher than the task being patched) would seem to have a similar chance of success.

Other alternatives:
- making a copy of the .o file on the ram disk, apply the patch there, and then use the app unload/load command to restart the app
- suspend the task, then apply the patch, then resume the task (still suffers a similar risk described above, but could feasibly work OK on SMP/multi-core setups, depends on knowing what task ID might execute the code in question).

skliper, 2020-07-29:
@acudmore @jwilmot - any of you know of this being used operationally?  I agree, sounds risky to patch executing code.  I've only done one flight patch and it definitely wasn't live.

jphickey, 2020-07-29:
At any rate, my proposal for a "path forward" here is:

- Remove "with interrupts disabled" text from requirement - as it is an implementation detail
- Recommend running MM command handler task with higher priority than other tasks.  This way, at least on a traditional single-core processor, the `memcpy` done by MM command will preempt all other tasks in the system.

These two things can be done easily without code change and will achieve mostly the same effect, thereby mitigating the effect of the `OS_IntLock()` removal.  Except for cases of timers / ISRs which will still preempt MM, but we don't have much code that executes in those contexts (it should quickly delegate to a normal task).  It also doesn't help SMP, but neither did OS_IntLock so its no different.

In the future, the ability to suspend/resume a task seems like a worthwhile addition, which would allow run to execute MM at a lower priority, and would work in SMP environments, preventing the task being patched from executing no matter what core it was assigned to.

**ALSO** - Worth noting that modern processors with MMUs may enforce permissions on memory pages too. Meaning that text/code pages are marked "execute only" -- and any attempt to read/write this program memory as data will result in a segfault.  That's a security feature, as it is common practice by malicious software to leverage string buffer overflow bugs to overwrite and modify the program code.

This is worth considering - it might not be worth investing much effort into supporting this approach of patching live code as it will probably be blocked by hardware on future flight systems.

skliper, 2020-08-06:
Near term approach from MM is to remove the intlock calls and update the related documentation/requirement.  A few use cases were discussed, and there wasn't a generic/abstract single solution since many times the trade involved operational approaches, various memory protection/permission impacts, SMP considerations, etc.  It can be re-evaluated in the future based on actual flight requirements/use cases/scenarios.

closed 2020-08-06 by skliper