nasa/PSP #202: No error message emitted when shmget() fails.

Jkillelea, 2020-10-04:
**Describe the bug**
On certain platforms, `shmget` may fail with `EINVAL` but the error handler does not call `OS_printf`. Interestingly enough, replacing the `OS_printf` call with regular `printf` worked and displays an error message.

This happened to me with a Raspberry Pi Zero when `CFE_PLATFORM_ES_USER_RESERVED_SIZE` was left at the default of `1024*1024`. This can be fixed by reducing the size to `1024*128`.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone cFS on an x86_64 host machine with ARM cross compilers (arm-linux-gnueabihf-gcc version 9.3.0).
2. Copy the build config files for cpu1 to cpu2 and modify `toolchain-cpu2.cmake` for ARM (change architecture and compiler paths).
3. Edit `targets.cmake` to compile for cpu2.
4. Build for cpu2, copy the directory `build/exe/cpu2` to the target machine
5. Run on the target machine (a Raspberry Pi Zero) with `sudo ./core-cpu2 --reset PO`
6. No error message is observed.

**Expected behavior**
cFS should either launch or fail with an error message.

**Code snips**
cFS exits with error code -1 but no error message.
```
CFE_PSP: Reset Type: PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 2
CFE_PSP: Default Spacecraft ID = 66
CFE_PSP: Default CPU Name: cpu2
OS_Posix_GetSchedulerParams():195:Policy 1: available, min-max: 1-99
OS_Posix_GetSchedulerParams():195:Policy 2: available, min-max: 1-99
OS_Posix_TaskAPI_Impl_Init():381:Selected policy 2 for RT tasks, root task = 99
pi@raspberrypi:~/cpu2 $ echo $?
255
```

**System observed on:**
 - Raspberry Pi Zero 
 - OS: Linux raspberrypi 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux
 - Versions: Current submodule versions listed in github.com/nasa/cFS on 4 Oct 2020.

**Reporter Info**
Jacob Killelea

Jkillelea, 2020-10-04:
Closing this for now because something else seems to be causing crashes as well, or is a deeper root cause. Still investigating.

skliper, 2020-10-05:
Marking as invalid for now.  If it's real we can reopen/recategorize.

jphickey, 2020-10-05:
Note that output using `OS_printf()` has an additional output buffer, and if the process exits immediately then there is a good chance the output never gets transferred from the output buffer to the console.

For situations where we are displaying a fatal error and will exit the process as a result, we should probably use `fprintf()` to `stderr` rather than the `OS_printf()` routine.  The latter is fine for normal status messages and errors/warnings which are not fatal.

jphickey, 2020-10-05:
Submitted #203 for the issue regarding `OS_printf()` message not appearing - I think this is a real bug.  Sounds like otherwise the Pi Zero has some limitations which require config tweaks - which is normal - but we should show the error message.

Jkillelea, 2020-10-06:
Interesting point about `OS_printf` having a buffer, I had forgotten and that's a good catch.

My `shmget` woes went away after a reboot. I've had older versions of cFS fail after a previous run had segfaulted, so I think it's likely related to that. Something about not cleaning up the CDS probably....

jphickey, 2020-10-07:
Yes, I've seen `shmget()` fail for a variety of reasons.  Note that these shared memory segments are a limited OS kernel resource and it can be possible to get stale segments laying around.  Sometimes you need to run `ipcrm` or just reboot to get back to normal.

Still, the point is that we really should be showing the error message when bad things happen, so if nothing else that's a bug.  (not the main bug, but still a bug).

skliper, 2020-10-07:
Reopened given the discussion.

jphickey, 2020-10-07:
@skliper - I had submitted the other issue #203 specifically to address the fact that the error message isn't displayed.  As to why `shmget()` failed in the first place there could be several possibilities - probably operational in nature (i.e. abnormal exits when debugging can leave stale segments, or something of that nature).  I'm not sure that we need a code change for it.

skliper, 2020-10-07:
Copy, thanks for the reminder.

closed 2020-10-07 by skliper