nasa/PSP #211: cES1503 requirement failure

dmknutsen, 2020-10-22:
**Describe the bug**
Current requirement verbiage:  Upon a Power-On Reset, the cFE shall clear the Volatile File system.

The cFE does not clear the volatile file system during a Power-On Reset.

**To Reproduce**
On a Linux System:
1. Place a file in the volatile file system  - /dev/shm/osal:RAM/
2. Start the software with -RPO (Power-On Reset) option. 
3. Verify that the file still exists.

**Expected behavior**
cFE should clear the Volatile File System or the requirement verbiage should be updated.

**System observed on:**
OS: ubuntu-19.10
Versions: cfe: v6.7.0+dev295; osal: v5.0.0+dev247; psp: v1.4.14.0

**Reporter Info**
Dan Knutsen
NASA/Goddard

skliper, 2020-10-22:
Transferred to PSP and attempted to update as suggested by CCB.  Open for comments/suggestions, and I suggest this caveat be included in test reports (and touched on in VDD).  We haven't done test reports for Linux in the past... but I'd like to get there (along with RTEMS and the typical VxWorks test report).

jphickey, 2020-10-22:
I would suggest if generating test reports against Linux targets, to do so by running CFE on a dev board like a Raspberry Pi or BBB, or within a VM which allows for a more flight-like setup.  

For instance, one can use the Yocto build system to generate a Linux base image for one of their example qemu-based targets (or RPi), then deploy CFE build to that system, and run the tests against that target.

Notably, this permits commands that require psp/system support, such as restart, to work correctly.

Running the cfe core executable directly from a command line on a dev box isn't really a valid-enough deployment scenario to test these more system-oriented commands.

skliper, 2020-10-22:
Actually we just discussed this today in a test meeting, and it would be fairly easy to fix even in the development environment when using CTF as long as a true power cycle isn't necessary (which it's not if the wrapper logic clears the volatile file system).   I'd rather not add complexity to the CTF test development at this point, since we benefit from keeping it as simple as possible.  Doing your suggestion from a real (or at least more realistic) testing perspective is a good goal to continue pursuing (along with RTEMS and VxWorks), it's just not a priority for the cert work.  Either way, lets retain this issue since it is a behavior on a dev box that will not meet this requirement and I think it's valuable to communicate this non-compliance.