nasa/PSP #244: Support VxWorks "muncher" for C++

jhnphm, 2021-01-28:
**Is your feature request related to a problem? Please describe.**
VxWorks requires a tool called a "muncher" in order to generate stubs required for static destructors/constructors to run, when building for C++. 

**Describe the solution you'd like**
Run the muncher. An example implementation is below:

In VxWorks-CFE.cmake, change 
```
set(CMAKE_CXX_CREATE_SHARED_MODULE ${CMAKE_C_CREATE_SHARED_MODULE})
```
to
```
if(NOT DEFINED CMAKE_CXX_CREATE_SHARED_MODULE)
    set(CMAKE_CXX_CREATE_SHARED_MODULE ${CMAKE_C_CREATE_SHARED_MODULE})
endif()
```
Somewhere in the target-specific PSP cmakefiles or toolchain file, define something like (the below is for SP0-S, derived from the workbench-generated makefiles)
```
set(VSB_DIR "${WIND_BASE}/target/lib")
set(VXMUNCHER_TAGS ${VSB_DIR}/tags/ppc/PPC85XX/e500v2common/dkm.tags)
set(VXMUNCHER_TCL "tclsh ${WIND_BASE}/host/resource/hutils/tcl/munch.tcl -c ppc -tags ${VXMUNCHER_TAGS}")
set(VXMUNCHER_COMPILE_FLAGS "-O2 -fstrength-reduce -te500v2 -mhard-float -fno-implicit-fp -mstrict-align -fno-builtin -fdollars-in-identifiers -c")
set(VX_LINK_FLAGS "-r -nostdlib -Wl,-X -T ${WIND_BASE}/target/h/tool/gnu/ldscripts/link.OUT")

set(CMAKE_CXX_CREATE_SHARED_MODULE
    "<CMAKE_CXX_COMPILER> <FLAGS> <CMAKE_C_LINK_FLAGS> ${VX_LINK_FLAGS} <LINK_FLAGS> <OBJECTS> -o <TARGET>.PartialImage.out <LINK_LIBRARIES>"
    "${CMAKE_NM} <TARGET>.PartialImage.out | ${VXMUNCHER_TCL} > <TARGET>.ctdt.c"
    "<CMAKE_C_COMPILER> -c <TARGET>.ctdt.c -o <TARGET>.ctdt.c.o ${VXMUNCHER_COMPILE_FLAGS}"
    "<CMAKE_CXX_COMPILER> <FLAGS> <CMAKE_C_LINK_FLAGS> <LINK_FLAGS> <TARGET>.PartialImage.out <TARGET>.ctdt.c.o -o <TARGET> ${VX_LINK_FLAGS}"
)
```

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**Requester Info**
John N Pham, Northrop Grumman

klystron78, 2021-01-28:
John, it's up to the user to figure out how to integrate cFS into their build environment. I don't think this is something we would consider since we aren't a c++ project either. Someone else correct me if I'm wrong.

skliper, 2021-01-28:
As @klystron78 mentioned, this would probably be more appropriate in a mission specific PSP (for the SP0 or whatever you are using). Right now our scope for the open source is still c (unless/until we get resources to expand support/testing to include c++).

jhnphm, 2021-01-28:
In that case, perhaps leave CMAKE_CXX_CREATE_SHARED_MODULE  unset if it's not supported so it can be overridden in the toolchain file/PSP ? Although I guess the PSP repo is for the most part cloned and owned so it's not a huge deal if left as-is.

skliper, 2021-02-12:
@mprather was the CMAKE_CXX_CREATE_SHARED_MODULE the issue for you also?