nasa/PSP #246: C++ comment style and commented out code violations

skliper, 2021-02-03:
**Is your feature request related to a problem? Please describe.**
C++ comment style and commented out code violates style guidelines
https://github.com/nasa/PSP/blob/0cd14abbb58691eb83afdd816e116f68e681b8e0/fsw/pc-rtems/src/cfe_psp_memory.c#L544-L565

**Describe the solution you'd like**
Clean up

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2021-02-03:
Need to figure out if these should be actually implemented or removed.

zanzaben, 2021-02-23:
The parts commented out have been that way since it was first brought over from the PSP 1.3.0.0 code base 2 years ago so I am confident it can be removed. 

Also the function CFE_PSP_MemReport isn't used anywhere else in the code base so should it be removed entirely.

astrogeco, 2021-02-24:
[CCB:2021-02-24](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-February-24-2021) APPROVED
- Remove, probably an old debug function

skliper, 2021-02-24:
CCB 20210224 - votes to remove function/file

PR #253 (2021-03-05): Fix #246, fix comment style
**Describe the contribution**
Fixes #246
Remove the unused code 

**Testing performed**
Build and run unit test

**Expected behavior changes**
No impact to behavior

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campbell GSFC
commit 07cd8614 (2021-03-05): Fix #246, remove unused code.
closed 2021-03-05 by skliper