nasa/PSP #251: Static Analysis: 'unsigned int' but is of type 'PCS_WIND_TCB *'

astrogeco, 2021-02-17:
**Describe the bug**
See <https://github.com/nasa/cFS/security/code-scanning/1?query=ref%3Arefs%2Fpull%2F193%2Fmerge>

Not sure we might want to dismiss it 

**To Reproduce**
Run CodeQL

**Code snips**
[image]

skliper, 2021-02-17:
It's not a cert issue, this would be open source resources please.

skliper, 2021-02-17:
Reference is out of date, if you use the static analysis from main: https://github.com/nasa/PSP/security/code-scanning/36?query=ref%3Arefs%2Fheads%2Fmain

and

https://github.com/nasa/PSP/blob/a1a7e7a3bfd6d2d82bb5d099c1f757ff60ba2431/fsw/mcp750-vxworks/src/cfe_psp_exception.c#L207-L208

anhvan989, 2026-02-11:
closing as duplicate. Issue #362

closed 2026-02-11 by anhvan989