nasa/PSP #252: Implement "CFE_PSP_GetVersion()" function

jphickey, 2021-02-18:
**Is your feature request related to a problem? Please describe.**
As described in nasa/osal#821 it is better to acquire version info strings/values at runtime rather than via a macro, because of where and how it will get evaluated.

PSP is actually in better shape than OSAL/CFE because `CFE_PSP_VERSION` currently refers to a string which is stored in GLOBAL_PSP_CONFIGDATA.  But still, it would be better and more conventional to get this value via an API call.

**Describe the solution you'd like**
Implement `CFE_PSP_GetVersion()` and related APIs to get the various version strings.

**Describe alternatives you've considered**
Leave in global object.

**Additional context**
See details in nasa/osal#821

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-03-24:
Implemented in #256, this is a duplicate

closed 2021-03-24 by jphickey

skliper, 2021-03-24:
Duplicate issue is #257