nasa/PSP #299: Add sleep(1) before exit call when printing

skliper, 2021-07-12:
**Is your feature request related to a problem? Please describe.**
User reported OS_print doesn't display before exit, for example:
https://github.com/nasa/PSP/blob/6d40816cfcc2836182f04bacf7d98b7d5cf5163b/fsw/pc-linux/src/cfe_psp_memory.c#L145-L146

**Describe the solution you'd like**
Add sleep(1) between prints and exits to give it a chance

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-07-14:
Similar/related issue to #203 (this is the symptom anyway).  I may even have a patch/fix that was never merged for that issue already done somewhere from when I quickly looked at it.

Note that use of `sleep` system call is discouraged for new code as it may interfere with other timers.  `nanosleep` (POSIX) or even OS_TaskDelay might be preferable.

pavll, 2021-08-29:
Ok, I now used the OS_TaskDelay since it's the implemented behaviour of the OSAL package but maybe I could use fflush(stdout) if someone who knows the project can give a short feedback on this. And if OS_TaskDelay is used maybe only 100 milliseconds are required because this value is already used in:
https://github.com/nasa/PSP/blob/6d40816cfcc2836182f04bacf7d98b7d5cf5163b/fsw/pc-linux/src/cfe_psp_start.c#L468-L474

PR #318 (2021-12-06): psp Integration candidate: Caelum+dev1
**Describe the contribution**

PR #301 from pavll/fix-299-sleep-before-exit

- Fix #299, Sleep before exit when printing

PR #314 from jphickey/fix-313-stub-databuf

- Fix #313, check the address in PSP get segment stubs

**Testing performed**
PSP Checks <https://github.com/nasa/PSP/pull/318/checks> 
Bundle Checks <https://github.com/nasa/cFS/pull/390/checks>

**Expected behavior changes**
See PRs

**System(s) tested on**
Ubuntu 18.04
RTEMS 4.11
RTEMS 5

**Additional context**
Part of <https://github.com/nasa/cFS/pull/390>

**Third party code**
None

**Co [... cut]
commit 20a2a34d (2021-12-06): Merge pull request #301 from pavll/fix-299-sleep-before-exit

Fix #299, Sleep before exit when printing
commit bf9b935b (2021-12-06): Fix #299, Sleep before exit when printing
PR #301 (2021-12-06): Fix #299, Sleep before exit when printing
**Describe the contribution**
- Fixes #299 

**Expected behavior changes**
- Print before exit [#299]

**Additional context**
- Considering if 100 milliseconds would be better than 1000 to save time and have a print.
- Considering fflush(stdout) instead of OS_TaskDelay

**Contributor Info - All information REQUIRED for consideration of pull request**
Paul Oberosler, Individual
closed 2021-12-06 by astrogeco