nasa/PSP #319: Module list count includes both `CFE_PSP_BASE_MODULE_LIST` and the configurable `PspModuleList` but some APIs use just the latter.

skliper, 2021-12-09:
**Describe the bug**
Both lists are initilized:
https://github.com/nasa/PSP/blob/c3a3f6ecaa33e62ded0d9b95dc91dd25b9e36a46/fsw/shared/src/cfe_psp_module.c#L82-L89

which increments the `CFE_PSP_ModuleCount`:
https://github.com/nasa/PSP/blob/c3a3f6ecaa33e62ded0d9b95dc91dd25b9e36a46/fsw/shared/src/cfe_psp_module.c#L72

but then the API's use `CFE_PSP_ModuleCount` to limit references to just the configurable list (and other related issues):
https://github.com/nasa/PSP/blob/c3a3f6ecaa33e62ded0d9b95dc91dd25b9e36a46/fsw/shared/src/cfe_psp_module.c#L120-L142

**To Reproduce**
For example if there's 1 entry in `CFE_PSP_BASE_MODULE_LIST` and 1 in the additional `PspModuleList`, then `CFE_PSP_ModuleCount` will == 2.  Then if you call CFE_PSP_MODULE_FindByName with any module not in the `PspModuleList` it'll actually go off the end of the `PspModuleList` since it's limited to 2... and it won't find anything in the `CFE_PSP_BASE_MODULE_LIST`.

**Expected behavior**
A complete module list should be used for APIs that includes both the built in and added.

**Code snips**
See above.

**System observed on:**
N/A - code inspection

**Additional context**
Stakeholder identified issue

**Reporter Info**
Jacob Hageman - NASA/GSFC

skliper, 2021-12-09:
Ping - @superolmo

jphickey, 2022-07-28:
Just saw this after I wrote up #353 (duplicate)

closed 2022-08-03 by dzbaker