nasa/PSP #384: Add example of hardware-based 1Hz signal

jphickey, 2023-03-28:
**Is your feature request related to a problem? Please describe.**
A few releases back the 1Hz configuration was changed to be initiated by CFE TIME itself, rather than started by the PSP.  This was done in order to resolve a race condition where the signal arrived before CFE TIME was ready to process it.

However, as a result, we no longer have a good example of a "real" hardware-based 1PPS/1Hz signal driving the CFE TIME 1Hz input, as the current example uses a software kernel timer only.

**Describe the solution you'd like**
Create an alternative to the `soft_timebase` module that reads a 1PPS signal from an external reference such as a serial port - as a number of GPS receivers do use a serial line to provide 1PPS along with location data.  This probably still wouldn't be directly usable but it would at least provide a (closer) example of how this could be achieved, that the user could modify for their particular setup.

**Describe alternatives you've considered**
There are two other ways this could be synced:
1. Use an external process like gpsd to sync the kernel clock to the 1PPS, then use `soft_timebase` as-is.  The result is that the 1Hz TIME signal will be in phase with the reference signal, but it won't occur at the same time.  This phase delay may or may not acceptable depending on the application.  This does have the advantage of having only a single time domain across the entire system (kernel/OS clocks and timers will all be synchronous with the GPS time) and it is totally transparent to apps - they just read time as normal.
2. Use an external sync routine (last argument to `OS_TimeBaseCreate` to wait for the 1PPS.  This is similar to 1 but easier to implement (no external dependencies) however the result will be more jittery and apps must know that this timebase may not be synchronous with the OS/kernel clock.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2023-03-28:
Marking as item for discussion

CDKnightNASA, 2024-01-04:
I think a RTC module for a Raspberry Pi would be a reasonable target for an example. Such as the I2C module: https://www.adafruit.com/product/3386#technical-details

This uses the RTC chip, which includes three timers (see section 8.9):
https://www.nxp.com/docs/en/data-sheet/PCF8523.pdf